analogdevicesinc/Linduino #1: The Wire.beginTransmission() and Wire.endTransmission() should not be used together with Wire.requestFrom().

Koepel, 2018-03-25:
In the file "Linduino/LTSketchbook/libraries/LT_I2C_Wire/LT_I2C_Wire.cpp", the function Wire.requestFrom() is often used together with Wire.beginTransmission() and Wire.endTransmission().

Regardless which Wire library or which processor is used, that is not correct.

Explanation: [Common-mistakes](https://github.com/Koepel/How-to-use-the-Arduino-Wire-library/wiki/Common-mistakes) number #2 and #3.

malavikasajikumar, 2018-03-26:
Thank you for pointing this out. We will have this fixed immediately.

Koepel, 2018-03-26:
The new [Adafruit_FT6206.cpp](https://github.com/adafruit/Adafruit_FT6206_Library/blob/master/Adafruit_FT6206.cpp) has it fixed. But there are also a few changes to the code. That might have influence on the rest of your code.

malavikasajikumar, 2018-06-11:
Fixed on Commit: 9e81d9d0069a4eb553788fa92f8ac9e21a583c4c

closed 2018-06-11 by malavikasajikumar