analogdevicesinc/Linduino #26: LTC2309, 'display_code' always 0

tnsolley, 2019-06-06:
Using the TSSOP variant of the LTC2309; Vdd, Vref, and REFCOMP have been connected using appropriate capacitors per page 8 of the [datasheet](https://www.analog.com/media/en/technical-documentation/data-sheets/2309fd.pdf) [as opposed-to mimicking the values used on the [DC1337A](https://www.analog.com/media/en/technical-documentation/eval-board-schematic/1337asch.pdf)].

I've tried using the `LTC2309.h`, `DC1337A.ino` codes in combination-with the Datasheet-caps and DC1337A-caps; all combinations have resulted in seeing a constant `0` value being printed from the output register.

Sample code below;
```
// Test ability of a single ADC chip, default address 

// Source code from LTC's Linduino / <DC1337A.ino> files.
/* 
 *  REFCOMP floating
 *  AD0 - GND
 *  AD1 - GND
 *  VREF - feed-to CH0
 *  COM - GND
 */

#include "LTC2309.h" 
#include "LT_I2C.h"

// Global variables
static uint8_t uni_bipolar = LTC2309_UNIPOLAR_MODE;    //!< LTC2309 Unipolar or Bipolar mode selection
static uint8_t single_ended_differential = LTC2309_SINGLE_ENDED_MODE;    //!< LTC2309 Unipolar or Bipolar mode selection
static uint8_t i2c_address = LTC2309_I2C_ADDRESS;       //!< I2C address in 7 bit format for part
static uint8_t LTC2309_bits = 12;                      //!< Resolution (12 bits)
static float LTC2309_vref = 4.096;

// Constants
// Build the command for single-ended mode
const uint8_t BUILD_COMMAND_SINGLE_ENDED[8] = {LTC2309_CH0, LTC2309_CH1, LTC2309_CH2, LTC2309_CH3,
    LTC2309_CH4, LTC2309_CH5, LTC2309_CH6, LTC2309_CH7
                                              };        //!< Builds the command for single-ended mode

void setup() {
  // put your setup code here, to run once:
  quikeval_I2C_init();                    // Enable the I2C port
  quikeval_I2C_connect();                 // Connect I2C to main data port
  Serial.begin(9600);
}

void loop() {
  // put your main code here, to run repeatedly:
  uint8_t user_command;
  uint16_t adc_command;         // The LTC2309 command byte
  uint16_t adc_code = 0;    // The LTC2309 code
  uint8_t x, y, startcount, endcount;
  uint16_t display_code;
  float adc_voltage;
  uint8_t acknowledge = 0;       // I2C acknowledge bit

  startcount = 0;
  endcount = 7;
  adc_command = BUILD_COMMAND_SINGLE_ENDED[0] | uni_bipolar;
  acknowledge |= LTC2309_read(LTC2309_I2C_ADDRESS, adc_command, &adc_code);     // Throws out last reading and starts CH0 conversion
  for (x = startcount; x <= endcount; x++)
  {
    adc_command = BUILD_COMMAND_SINGLE_ENDED[x % 8] | uni_bipolar; // Send channel config for the NEXT conversion to take place
    acknowledge |= LTC2309_read(LTC2309_I2C_ADDRESS, adc_command, &adc_code);     // Throws out last reading and starts CH0 conversion
    acknowledge |= LTC2309_read(LTC2309_I2C_ADDRESS, adc_command, &adc_code);   // Read previous channel conversion (x-1) and start next one (x)

    display_code = adc_code >> (16 - LTC2309_bits);
    display_code = display_code & 0xFFF;

    Serial.print(F("  Received Code: b"));
    Serial.println(display_code, BIN);

    adc_voltage = LTC2309_code_to_voltage(adc_code, LTC2309_vref, uni_bipolar);
    Serial.print(F("  Voltage read on "));
    Serial.print(F("Ch"));
    Serial.print(x);
    Serial.print(F(": "));
    Serial.print(adc_voltage, 4);
    Serial.println(F("V"));
    Serial.println();
  }
}
// EOL
```

tnsolley, 2019-06-06:
Follow-up; attempting to interface with the LTC2309 using an Arduino Uno. PC is running Windows 10.

malavikasajikumar, 2019-06-06:
Thanks for reaching out. Could you provide a schematic of your circuit and oscilloscope scope photos showing SDA and SCL. Also what channel are you trying to convert and what voltage is being applied to the ADC inputs? Please send the details to linduino@analog.com.

tnsolley, 2019-06-06:
Will do! I was just about to also post some additional info here.

After running into this issue with the LTC2309 library, I did try to use standard `Wire.h` functions, with, I'm surprised to say, some limited success. I could get data from each Channel, and these values did appear to respond to the input... However, data on Ch.0 was not scaling properly, and I've since been unable to figure what segment from the `LTC2309.cpp` I'm missing to account-for this.

Sample code of my Wire attempt, for those interested;
```
// See <adc_singlechipTest1.ino> for ref code and configs
// Caps configured per page 8 of the LTC2309 datasheet

/*
 * Attempting to use the standard Wire.h library, in place-of <LTC2309.h>
 */

#include <Wire.h>

#define devAdr0 0x08 // I2C address
#define devAdrA 0x1A
#define devAdrB 0x14
#define comReg 0x11   // command register
#define dinWord0 0x80 // 6-bit, S/D-O/S-S1-S0-UNI-SLP = 1000-0-0-xx ? (0x80)
#define dinWord1 0xC0 // 1100-0000
#define dinWord2 0x90 // 1001-0000
#define dinWord3 0xD0 // 1101-0000
#define dinWord4 0xA0 // 1010
#define dinWord5 0xE0 // 1110
#define dinWord6 0xB0 // 1011
#define dinWord7 0xF0 // 1111

int reading = 0;

float readdata(int chip, int channel){
  uint8_t devAdr;
  uint8_t dinWord;

  // Assign
  if(chip == 0){
    devAdr = devAdr0;
  }

  if(channel == 0){
    dinWord = dinWord0;
  }
  else if(channel == 1){
    dinWord = dinWord1;
  }
  else if(channel == 2){
    dinWord = dinWord2;
  }
  else if(channel == 3){
    dinWord = dinWord3;
  }
  else if(channel == 4){
    dinWord = dinWord4;
  }
  else if(channel == 5){
    dinWord = dinWord5;
  }
  else if(channel == 6){
    dinWord = dinWord6;
  }
  else if(channel == 7){
    dinWord = dinWord7;
  }
  
  Wire.beginTransmission(devAdr);
  //Wire.write(byte(comReg));         // Send the Address plus Read cmd.
  Wire.write(byte(dinWord));          // Program it, for Ch0. 0 single ended unipolar
  Wire.endTransmission();

  // Request
  Wire.requestFrom(devAdr, 2);    // Request 2 byte

  // Read
  if(2 <= Wire.available()){
    reading = Wire.read();    // First byte; clear prior output
    reading = reading << 8;
    reading |= Wire.read(); // Grabs second byte, appends to first

    // Maybe if I parse-out the second byte?
    
    // Steal-from LTC2309, <LTC2309_code_to_voltage()>
    float voltage;
    voltage = (float)reading;
    voltage = voltage / (pow(2,16)-1);
    voltage = voltage * 4.096;
    return voltage;
  }
}

void setup() {
  // put your setup code here, to run once:
  Serial.begin(9600);
  Wire.begin();
}

void loop() {
  int cn = 0;
  // put your main code here, to run repeatedly:
  for(int i=0; i<8; i++){
    float voltage = readdata(0,i);
    Serial.print("Chip ");
    Serial.print(cn);
    Serial.print(", Channel ");
    Serial.print(i);
    Serial.print(", Voltage ");
    Serial.println(voltage);
    delay(250);
  }  
}
// EOL
```

tnsolley, 2019-06-13:
For those curious, this was a case of several things all going wrong;
1) The Uno does not have internal pullup resistors on SDA or SCL, so 4.7k Ohm resistors have to be added on each of those lines.
2) Address selection. When using multiple chips, refer to Table 2 of the datasheet for proper address inits.

Working code for a single chip, Arduino Uno, IDE 1.8.8;
```
// Test ability of a single ADC chip, user-specified address 

// Source code from LTC's Linduino / <DC1337A.ino> files.
/* 
 *  REFCOMP floating
 *  AD0 HIGH
 *  AD1 GND
 *  VREF floating
 *  COM GND
 */

#include "LTC2309.h" 
#include "LT_I2C.h"
#define dev_addr 0x0A
#define chipA_pin 2

// Global variables
static uint8_t uni_bipolar = LTC2309_UNIPOLAR_MODE;    //!< LTC2309 Unipolar or Bipolar mode selection
static uint8_t single_ended_differential = LTC2309_SINGLE_ENDED_MODE;    //!< LTC2309 Unipolar or Bipolar mode selection
//static uint8_t i2c_address = LTC2309_I2C_ADDRESS;       //!< I2C address in 7 bit format for part
static uint8_t i2c_address = dev_addr;       //!< I2C address in 7 bit format for part
static uint8_t LTC2309_bits = 12;                      //!< Resolution (12 bits)
static float LTC2309_vref = 4.096;

// Constants
// Build the command for single-ended mode
const uint8_t BUILD_COMMAND_SINGLE_ENDED[8] = {LTC2309_CH0, LTC2309_CH1, LTC2309_CH2, LTC2309_CH3,
    LTC2309_CH4, LTC2309_CH5, LTC2309_CH6, LTC2309_CH7
                                              };        //!< Builds the command for single-ended mode

void setup() {
  // put your setup code here, to run once:
  pinMode(chipA_pin, OUTPUT);
  quikeval_I2C_init();                    // Enable the I2C port
  quikeval_I2C_connect();                 // Connect I2C to main data port
  Serial.begin(9600);
}

void loop() {
  // put your main code here, to run repeatedly:
  digitalWrite(chipA_pin, HIGH);
  uint8_t user_command;
  uint16_t adc_command;         // The LTC2309 command byte
  uint16_t adc_code = 0;    // The LTC2309 code
  uint8_t x, y, startcount, endcount;
  uint16_t display_code;
  float adc_voltage;
  uint8_t acknowledge = 0;       // I2C acknowledge bit

  //startcount = 0;
  //endcount = 7;
  startcount = 0;
  endcount = 0;
  adc_command = BUILD_COMMAND_SINGLE_ENDED[0] | uni_bipolar;
  acknowledge |= LTC2309_read(LTC2309_I2C_ADDRESS, adc_command, &adc_code);     // Throws out last reading and starts CH0 conversion
  for (x = startcount; x <= endcount; x++)
  {
    adc_command = BUILD_COMMAND_SINGLE_ENDED[x % 8] | uni_bipolar; // Send channel config for the NEXT conversion to take place
    acknowledge |= LTC2309_read(dev_addr, adc_command, &adc_code);     // Throws out last reading and starts CH0 conversion
    acknowledge |= LTC2309_read(dev_addr, adc_command, &adc_code);   // Read previous channel conversion (x-1) and start next one (x)

    display_code = adc_code >> (16 - LTC2309_bits);
    display_code = display_code & 0xFFF;

    Serial.print(F("  Received Code: b"));
    Serial.println(display_code, BIN);

    adc_voltage = LTC2309_code_to_voltage(adc_code, LTC2309_vref, uni_bipolar);
    Serial.print(F("  Voltage read on "));
    Serial.print(F("Ch"));
    Serial.print(x);
    Serial.print(F(": "));
    Serial.print(adc_voltage, 4);
    Serial.println(F("V"));
    Serial.println();
  }
  delay(100);
  digitalWrite(chipA_pin, LOW);
  delay(1000);
}
// EOL
```

closed 2019-06-13 by tnsolley