analogdevicesinc/Linduino #35: Fix for compiler warning in LTC681x.cpp

GunZi200, 2020-02-08:
In line 665 in LTC681x.cpp this line gives a compiler warning for undefined behaviour:

`parsed_stat = data[data_counter++] + (data[data_counter++]<<8);`

Changing this to:

```
parsed_stat = data[data_counter] + (data[data_counter+ 1]<<8);
data_counter = data_counter + 2;
```

resolves the warning.

jothishjoseph, 2020-03-20:
Hope it is tested and working fine, approving the change

GunZi200, 2020-03-24:
Just to make things more clear, this is what the compiler warning looks like on my end using the library before this commit. 
[image]
I basically copied the code in the if statement above which does the same thing but without any warnings from the compiler.

[image]

The  `if (reg == 2) `statement is where I am suggesting changes.